l1npengtul/nokhwa #177: Cannot use nokhwa because of mozjpeg-sys deprecation

ktecho, 2024-07-24:
I'm trying to use nokhwa, but branch 1.1 of `mozjpeg-sys` has been deprecated:

https://github.com/kornelski/mozjpeg-sys/issues/39

So I'm getting this:

```
error: failed to select a version for the requirement `mozjpeg-sys = "^1.1"`
candidate versions found which didn't match: 2.2.0, 2.1.0, 2.0.6, ...
location searched: crates.io index
required by package `mozjpeg v0.9.8`
    ... which satisfies dependency `mozjpeg = "^0.9"` of package `nokhwa-core v0.1.2 (https://github.com/l1npengtul/nokhwa.git?branch=0.10#c5a56bd2)`
    ... which satisfies git dependency `nokhwa-core` (locked to 0.1.2) of package `nokhwa v0.10.4 (https://github.com/l1npengtul/nokhwa.git?branch=0.10#c5a56bd2)`
    ... which satisfies git dependency `nokhwa` (locked to 0.10.4) of package `resurrection_wallet v0.0.1 (/home/ktecho/RustroverProjects/resurrection_wallet/src-tauri)`

```

What can we do?

kornelski, 2024-07-26:
This version will **stop working forever** starting with Rust 1.81, even if you have it in the lockfile.

Skgland, 2024-08-06:
I am currently working around the absence of a patched release by patching my binary crate in Cargo.toml

```Cargo.toml

[patch.crates-io]
# patch nokhwa core until https://github.com/l1npengtul/nokhwa/pull/178 is merged and released
nokhwa-core = { git = "https://github.com/l1npengtul/nokhwa.git", rev = "74a98ace10368320cfcc2186118ae10d5308ca59"}
```

kendfrey, 2024-09-02:
I've added the patch to my fork, and also updated the `image` dependency to `0.25`. I'm using this dependency in my Cargo.toml:
```toml
nokhwa = { version = "0.10.4", features = ["input-native", "output-threaded"], git = "https://github.com/kendfrey/nokhwa.git", branch = "0.10" }
```
This fixes the mozjpeg issue and the out-of-date `image` dependency. Also, the reason I'm using `output-threaded` is because the `Send` trait on `Camera` requires it (as of this version).

marcempunkt, 2024-10-07:
Quick question: kendfrey's fork works but is this going to be fixed? :slightly_smiling_face:

l1npengtul, 2024-10-12:
Ill try to push out 0.10.5, and remove the sync requirement on output threaded.

l1npengtul, 2024-10-12:
0.10.5 has been published. Please feel free to make a new issue if the issue persists.

commit fafffac4 (2024-10-12): update dependencies fix #177
closed 2024-10-12 by l1npengtul