l1npengtul/nokhwa #184: Corrupted image

Vanello1908, 2024-11-09:
```
pub fn create_photo(path: &PathBuf) -> Result<(), &'static str>{
    let index: CameraIndex = CameraIndex::Index(0); 
    let requested: RequestedFormat<'_> = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestResolution);
    let res: Result<Camera, nokhwa::NokhwaError> = Camera::new(index, requested);
    if res.is_err(){
        return Err("Camera error");
    }
    let mut camera: Camera = res.unwrap();
    let frame = camera.frame().unwrap();
    let decoded = frame.decode_image::<RgbFormat>().unwrap();
    
    match decoded.save_with_format(path, image::ImageFormat::Jpeg){
        Ok(_) => {},
        Err(_) => return Err("Photo write error")
    }
    Ok(())
}
```
But when I shot something fully black from one laptop, I have
[image]
But from another
[image]

Vanello1908, 2024-11-09:
I've tried different methods:
- changing decoder
- entering "closest" format
- absolute highest framerate
- entering "exact" format
- changing output type

l1npengtul, 2024-12-03:
Did you try opening the camera stream before calling `frame()`?