l1npengtul/nokhwa #186: crates.io publish request

darioalessandro, 2024-11-28:
Hello @l1npengtul :

Thank you for building this awesome tool! 

I want to release `video-call-daemon` to crates.io

I depend on the latest commit of https://github.com/l1npengtul/nokhwa/tree/0.10

Would you kindly publish this crate to crates.io so that I can release?

Thanks

Dario

l1npengtul, 2024-11-28:
I will test it tonight and release very soon. Please keep tuned for 0.11!

l1npengtul, 2024-11-28:
Probably, although macos will be a "maybe" (no mac machine)

darioalessandro, 2024-11-28:
I can volunteer to test. I have 4 macbook computers at your service:

2 intel: i5 and i7
2 arm: M1 and M2

How can I help you to validate the crate? Would you have an integration test?

[image]

darioalessandro, 2024-11-30:
How are things looking? you mentioned that macos might be the blocker, can I help you with that?

darioalessandro, 2024-12-03:
Hey @l1npengtul how are things looking?

l1npengtul, 2024-12-03:
> Hey @l1npengtul how are things looking? 

please give me a little bit of time... school is starting to get busy 😔

l1npengtul, 2024-12-03:
0.10.6 published, closing.

closed 2024-12-03 by l1npengtul

darioalessandro, 2024-12-04:
@l1npengtul thank you so much!!!