l1npengtul/nokhwa #192: nokwha::query(ApiBackend::MediaFoundation) causes panic on Windows

njrogie, 2024-12-22:
On windows, desktop PC, no builtin camera, I ran the following code:

```rust
let cameras = nokhwa::query(ApiBackend::MediaFoundation);
match cameras {
    Err(err) => {
        info!("No cameras present!");
    }, 
    Ok(cameras) => {
         for (idx, camera) in cameras.iter().enumerate() {
              info!("{}: {}", idx, camera.human_name())
         }
    }
}
```

When i plug my usb webcam in, i get the following output:
```
0: NexiGo N60 FHD Webcam
```

When no camera is plugged in, the windows bindings panic.
```
thread 'tests::test_print_info' panicked at core\src\panicking.rs:221:5:
unsafe precondition(s) violated: slice::from_raw_parts requires the pointer to be aligned and non-null, and the total size of the slice not to exceed `isize::MAX`
stack backtrace:
   0: std::panicking::begin_panic_handler
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library/std\src\panicking.rs:662
   1: core::panicking::panic_nounwind_fmt::runtime
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library/core\src\panicking.rs:112
   2: core::panicking::panic_nounwind_fmt
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library/core\src\panicking.rs:122
   3: core::panicking::panic_nounwind
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library/core\src\panicking.rs:221
   4: core::slice::raw::from_raw_parts::precondition_check
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library\core\src\ub_checks.rs:70
   5: core::slice::raw::from_raw_parts<enum2$<core::option::Option<windows::Windows::Win32::Media::MediaFoundation::IMFActivate> > >
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library\core\src\ub_checks.rs:77
   6: nokhwa_bindings_windows::wmf::query_activate_pointers
             at C:\Users\njrog\.cargo\registry\src\index.crates.io-6f17d22bba15001f\nokhwa-bindings-windows-0.4.1\src\lib.rs:278
   7: nokhwa_bindings_windows::wmf::query_media_foundation_descriptors
             at C:\Users\njrog\.cargo\registry\src\index.crates.io-6f17d22bba15001f\nokhwa-bindings-windows-0.4.1\src\lib.rs:365
   8: nokhwa::query::query_msmf
             at C:\Users\njrog\.cargo\registry\src\index.crates.io-6f17d22bba15001f\nokhwa-0.10.5\src\query.rs:269
   9: nokhwa::query::query
             at C:\Users\njrog\.cargo\registry\src\index.crates.io-6f17d22bba15001f\nokhwa-0.10.5\src\query.rs:98
  10: display_video::impl$0::default
             at .\src\lib.rs:12
  11: display_video::tests::test_print_info
             at .\src\lib.rs:39
  12: display_video::tests::test_print_info::closure$0
             at .\src\lib.rs:38
  13: core::ops::function::FnOnce::call_once<display_video::tests::test_print_info::closure_env$0,tuple$<> >
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library\core\src\ops\function.rs:250
  14: core::ops::function::FnOnce::call_once
             at /rustc/f6e511eec7342f59a25f7c0534f1dbea00d01b14\library/core\src\ops\function.rs:250
```

Seems like the windows bindings may need to be updated to handle a 'no-camera' state?

njrogie, 2024-12-22:
Whoops i see now that this issue was fixed months ago, i'm not sure why the version of nokwha i installed didn't have that fix in it. investigating

njrogie, 2024-12-22:
On returning to this today, I'd really like to be able to compile from source in windows to be able to use the most recent code. However, the build for `nokhwa-core` is broken at the moment (errors attached). Is there a correct way to get the build to work?`
[nokhwa-build.txt](https://github.com/user-attachments/files/18223419/nokhwa-build.txt)

l1npengtul, 2024-12-23:
`senpai` is broken, have you tried using `cargo update` and using nokhwa 0.10.7?

njrogie, 2024-12-28:
Here's my cargo.toml (the project is to try to display the camera feed in `egui`, which isn't quite relevant to this issue but it's worth mentioning).

```toml
[package]
name = "display-video"
version = "0.1.0"
edition = "2021"

[dependencies]
# egui
eframe = { version = "0.30.0", features = ["default"] }
epaint = { version = "0.30.0" }

# nokhwa  
nokhwa = { version = "0.10.7", features = ["input-msmf"]}

# Logging
env_logger = { version = "0.11.5", default-features = false, features = [
    "auto-color",
    "humantime",
] }
log = "0.4.22"
```

after running `cargo update` I still have the older `nokwha-bindings-windows` stored in `.cargo\registry\src` and `.cargo\registry\cache`. So i deleted the stored crate on my computer and re-downloaded it. The downloaded version from `crates.io` for `nokwha-bindings-windows 0.4.2` continues to have the bug in it, even when the `nokwha` version is set to 0.10.7.

njrogie, 2024-12-28:
Also, I wanted to know whether you would be open to a PR for fixing the `senpai` build?