l1npengtul/nokhwa #203: No output format support

TrevorBlythe, 2025-02-22:
Trying to save an image from the camera with 
```
frame.save(Path::new("output.jpg")).unwrap();
```
yeilds 
```
 on an `Err` value: Unsupported(UnsupportedError { format: Unknown, kind: Format(Name("Jpeg")) })
```

Every other type ive tried failed to.

l1npengtul, 2025-02-22:
By default, the image crate is built in `default-features=false`

Ensure you have all the features required for JPEG encoding enabled in the version of the image crate you are using.

Feel free to reopen the issue if you encounter more problems.

closed 2025-02-22 by l1npengtul

TrevorBlythe, 2025-02-23:
Even with all these features it still dont to work
```
[dependencies]
image = { version = "0.24.5", features = ["png"] , default-features = true}

[dependencies.nokhwa]
version = "0.10.0"
default-features = true
features = ["input-native", "output-wgpu", "serialize", "default", "output-threaded"]
``` 
Ive tried with default features on off for both image and nokhwa 😢

l1npengtul, 2025-02-23:
Try removing the patch specifier for the image crate, and adding the jpeg decoder feature, try cargo update as well.

TrevorBlythe, 2025-02-24:
Yo thanks it worked for anyone else that comes across this problem this worked for me.

[dependencies]
image = { version = "0.25", features = ["png", "jpeg"]}

[dependencies.nokhwa]
version = "0.10.0"
features = ["input-native", "output-wgpu", "serialize", "default", "output-threaded"]

l1npengtul, 2025-02-24:
> Yo thanks it worked for anyone else that comes across this problem this worked for me.
> 
> [dependencies] image = { version = "0.25", features = ["png", "jpeg"]}
> 
> [dependencies.nokhwa] version = "0.10.0" features = ["input-native", "output-wgpu", "serialize", "default", "output-threaded"]

You probably also want to remove the patch specifier for nokhwa, there is a new 0.10.7 patch out that fixes some issues you may encounter.