l1npengtul/nokhwa #204: Hangs and errors on MacOS

brookman, 2025-02-25:
Hi there!

I'm trying to get a camera image from the built-in camera of my MacBook Pro M2 like this:

```rust
    let backend = native_api_backend().context("Error with creating native camera backend")?;
    let devices = query(backend)?;
    println!("There are {} available cameras.", devices.len());
    for device in devices {
        println!("{device}");
    }

    let requested = RequestedFormat::new::<RgbFormat>(RequestedFormatType::AbsoluteHighestResolution);
    let mut camera = Camera::new(CameraIndex::Index(0), requested)?;
```

If I use `RequestedFormatType::AbsoluteHighestResolution` or `RequestedFormatType::AbsoluteHighestResolution`, Camera::new hangs forever.

If I use `RequestedFormatType::None` I'm getting this error:

```
2025-02-25 15:20:25.172819+0100 webcam[53629:13618507] [plugin] AddInstanceForFactory: No factory registered for id <CFUUID 0x6000038819e0> F8BB1C28-BAE8-11D6-9C31-00039315CD46
There are 1 available cameras.
Name: FaceTime HD Camera, Description: Apple Inc.: FaceTime HD Camera - AVCaptureDeviceTypeBuiltInWideAngleCamera, Unspecified f0, Extra: 3F45E80A-0176-46F7-B185-BB9E2C0E82E3, Index: 0
Error: Could not set device property CameraFormat with value 1920x1080@15FPS, YUYV Format: Not Found/Rejected/Unsupported

Stack backtrace:
   0: std::backtrace_rs::backtrace::libunwind::trace
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/../../backtrace/src/backtrace/libunwind.rs:116:5
   1: std::backtrace_rs::backtrace::trace_unsynchronized
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2: std::backtrace::Backtrace::create
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/backtrace.rs:331:13
   3: anyhow::error::<impl core::convert::From<E> for anyhow::Error>::from
             at /Users/beba/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/anyhow-1.0.96/src/backtrace.rs:27:14
   4: <core::result::Result<T,F> as core::ops::try_trait::FromResidual<core::result::Result<core::convert::Infallible,E>>>::from_residual
             at /Users/beba/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/result.rs:2014:27
   5: webcam::main
             at ./webcam/src/main.rs:44:22
   6: core::ops::function::FnOnce::call_once
             at /Users/beba/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/ops/function.rs:250:5
   7: std::sys::backtrace::__rust_begin_short_backtrace
             at /Users/beba/.rustup/toolchains/stable-aarch64-apple-darwin/lib/rustlib/src/rust/library/std/src/sys/backtrace.rs:152:18
   8: std::rt::lang_start::{{closure}}
   9: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/core/src/ops/function.rs:284:13
  10: std::panicking::try::do_call
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/panicking.rs:584:40
  11: std::panicking::try
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/panicking.rs:547:19
  12: std::panic::catch_unwind
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/panic.rs:358:14
  13: std::rt::lang_start_internal::{{closure}}
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/rt.rs:174:48
  14: std::panicking::try::do_call
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/panicking.rs:584:40
  15: std::panicking::try
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/panicking.rs:547:19
  16: std::panic::catch_unwind
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/panic.rs:358:14
  17: std::rt::lang_start_internal
             at /rustc/4d91de4e48198da2e33413efdcd9cd2cc0c46688/library/std/src/rt.rs:174:20
  18: _main
```

zyx-db, 2025-03-04:
Getting the same issue on macOS Sonoma 14.4.1

ungerpeter, 2025-03-19:
I am using a subsequent revision, and that code works on my MacBook Pro M2, could you try it?

612c861ef153cf0ee575d8dd1413b960e4e19dd6

zyx-db, 2025-03-19:
Im able to get that revision working on my computer (MacBook Air M1, MacOS Sonoma Version 14.4.1)

Thank you for sharing!