l1npengtul/nokhwa #214: 0.18.0 is broken?

marc2332, 2025-07-05:
Why was a breaking change released in a patch release? And for that reason the macOS (and maybe the others) impl also breaks?

https://github.com/marc2332/freya/actions/runs/16086291334/job/45397823112#step:8:144

[image]

l1npengtul, 2025-07-06:
Thats what I get for pushing without testing 😭

l1npengtul, 2025-07-06:
> @l1npengtul I see you yanked 0.10.8 but you nokhwa-core still has the 0.1.6 release, maybe yank that one as well :)

I just have...
Im setting up a macos and windows vm... hopefully I'll have it fixed snd republished by the end of day

marc2332, 2025-07-06:
Sure, take your time!

When you release could you make it as `0.11.0` instead of a patch release ?

Also, I see that you have a CI where all the worksflows fail, maybe use that as a reference to see if its failing or not, no? Do you maybe need help with that?

l1npengtul, 2025-07-06:
> When you release could you make it as `0.11.0` instead of patch releases ?

noooo but 0.11 is the Big Shiny Update That Will Fix All Our Problems:tm: ive been working on in senpai branch

> Also, I see that you have a CI where all the worksflows fail, maybe use that as a reference to see if its failing or not, no? Do you maybe need help with that?

uhh ill get to those eventually

l1npengtul, 2025-07-06:
ohhhhhh im stupiddddddddd
the code works its just that i forgor!! to publish the macos and linux crate

l1npengtul, 2025-07-06:
me when i dont read 

ill make sure the windows version compiles and push the changes to crates

marc2332, 2025-07-06:
Maybe you could yank the windows bindings release for now too

marc2332, 2025-07-06:
> ive just built the entire thing on all 3 platforms, it builds fine (albeit with a "few" warnings :p)

Sure but I would do this if I was you:
- Yank the windows release (that's the one missing from release gone wrong)
- And then only release the changes you have been cooking in a 0.11.0 release so that 0.10 keeps working

l1npengtul, 2025-07-06:
> Sure but I would do this if I was you:
> - Yank the windows release (that's the one missing from release gone wrong)
> - And then only release the changes you have been cooking in a 0.11.0 release so [...]

i havent backported any changes from 0.11, it should still work, and ive fixed compile issues from newer versions of wgpu

l1npengtul, 2025-07-06:
ok, ive republished the crates, please open another issue if its broken again

closed 2025-07-06 by l1npengtul